google/globalfoundries-pdk-libs-gf180mcu_fd_pr #5: Improve the top level spice file names under models

mithro, 2022-07-22:
The names of the top level spice files are currently;
```
./models
./models/ngspice
./models/ngspice/design.ngspice
./models/ngspice/sm141064.ngspice
./models/ngspice/smbb000149.ngspice
./models/xyce
./models/xyce/design.xyce
./models/xyce/sm141064.xyce
./models/xyce/smbb000149.xyce
```

The names `design` / `sm141064` / `smbb000149` don't really have useful meanings. I believe that @mkkassem has better names that should be used instead.

mithro, 2022-07-22:
I think @mkkassem was suggesting something like this?
 * `design` --> `gf180mcu.global_parameters`
 * `sm141064` --> `gf180mcu.common`
 * `smb000149` --> `gf180mcu.highvoltage`

@mkkassem - Can you document what these files should be called and then assign the issue to the right person to fix?

atorkmabrains, 2022-07-25:
@mithro I agree with the suggestion above. If @mkkassem approves, we could move forward.

mohanad0mohamed, 2022-08-07:
@mithro Should I start to fix the issue with @mkkassem suggestions?