google/globalfoundries-pdk-libs-gf180mcu_fd_pr #15: Make sure the DRC rules have unit tests as well as bigger integration tests

mithro, 2022-08-03:
It is general best practice to have lots of small units which only test one thing at a time. At the moment, the testing in #11 are more like integration tests.

The testing for the DRC rules should look like this -- for each DRC rule, there should be a directory which contains;
 * A human readable description of the rule
 * A set of GDS files which demonstrate different violations which should be caught by the DRC rule. These should also be rendered as examples into the human readable description of the rule.
 * A set of GDS files which should not have any violations in them.
 * These GDS test cases should be run with *only* the individual DRC rule being enabled.

There should **still** be DRC integration tests, but they should only be run after all the individual unit tests are successful.

atorkmabrains, 2022-08-07:
@mithro The DRC tests are unit tests for each rule. Unfortunately, it's hard to isolate each rule as the DRc runs takes long time. All tests are integrated one GDS files and we could tell which rule had missed errors or false positive violations. There some parts of the regression system that are not integrated yet. @mohanad0mohamed will integrate them soon that detects each rule on its own. But it will be single run for most of the rules.

mithro, 2022-08-10:
We should split the GDS file apart so that each test has multiple small GDS files demonstrating one thing at a time.

atorkmabrains, 2022-08-10:
@mithro As we have discussed, we will split the GDS testing file to smaller test units per rule to be able to parallelize the run.