google/globalfoundries-pdk-libs-gf180mcu_fd_pr #21: Shouldn't use `os.system` in Python scripts

mithro, 2022-08-10:
## Expected Behavior

You shouldn't use [`os.system`](https://docs.python.org/3/library/os.html#os.system) in Python scripts.

Examples;

Should use python internal functionality;

https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/2e2eb4ad16e9f26ae80f7b0cad70dfd9a10da177/rules/klayout/lvs/testing/run_sc_regression.py#L105

https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/2e2eb4ad16e9f26ae80f7b0cad70dfd9a10da177/rules/klayout/lvs/testing/run_sc_regression.py#L139

Should use `subprocess` module;

https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/2e2eb4ad16e9f26ae80f7b0cad70dfd9a10da177/rules/klayout/lvs/testing/run_sc_regression.py#L138

## Actual Behavior

You should use inbuilt Python functions or the [`subprocess` module](https://docs.python.org/3/library/subprocess.html) like [`check_call`](https://docs.python.org/3/library/subprocess.html?highlight=check_call#subprocess.check_call) or [`check_output`](https://docs.python.org/3/library/subprocess.html?highlight=check_call#subprocess.check_output).

Instead of using `rm` you should use something like [`Path.unlink`](https://docs.python.org/3/library/pathlib.html?highlight=unlink#pathlib.Path.unlink) or [`os.unlink`](https://docs.python.org/3/library/os.html?highlight=unlink#os.unlink) or [`shutil.rmtree`](https://docs.python.org/3/library/shutil.html?highlight=shutil#shutil.rmtree).

mithro, 2022-08-10:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/2e2eb4ad16e9f26ae80f7b0cad70dfd9a10da177/rules/klayout/lvs/testing/run_sc_regression.py#L226

Should probably use https://docs.python.org/3/library/shutil.html?highlight=shutil#shutil.move or https://docs.python.org/3/library/os.html#os.rename or https://docs.python.org/3/library/os.html#os.replace

atorkmabrains, 2022-08-10:
@mohanad0mohamed Could you please take care of this one soon?

FaragElsayed2, 2023-04-09:
@atorkmabrains This issue has been addressed in [PV repo ](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pv)

atorkmabrains, 2023-04-09:
This has been addressed in PV repo as highlighted by @FaragElsayed2 

@mithro I'll close this issue. Please don't hesitate to open if you have any other concerns.

closed 2023-04-09 by atorkmabrains