google/globalfoundries-pdk-libs-gf180mcu_fd_pr #30: Create a custom image to use for our CI testing

atorkmabrains, 2022-08-10:
## Need to have a custom ubuntu image to avoid the need to re-install klayout version 0.27.8 every time.

@umarcor Is is possible to help us and create a custom image that we could use it for our CI instead of installing klayout every time?

Actually, you could install all the tools needed using the script here:
https://github.com/mabrains/OpenAnalogDesign/blob/main/docker_build/Makefile

Also, you will need to get the file below next to that Makefile for the installation to complete:
https://github.com/mabrains/OpenAnalogDesign/blob/main/docker_build/cmake_init.sh

You will need to change klayout version in that file to 0.27.8 instead of 0.27.10.

umarcor, 2022-08-10:
There is an image from hdl/containers available already. Alternatively, we could build one in this repo and use the build-in container registry. Nevertheless, let's deal with #29, #28 and #18 first, since #18 will make it easier to implement this issue.

atorkmabrains, 2022-08-10:
@umarcor I already did that here: https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/27

atorkmabrains, 2022-08-10:
No need for #29

atorkmabrains, 2022-08-10:
Could we also wait for #18 and #28 as we will split the test cases further as mentioned in #15 ?

atorkmabrains, 2022-08-10:
@umarcor I think having this image would be really helpful. I don't know how to do it. It takes about 7 mins just to install klayout and there will be other tools.

atorkmabrains, 2022-08-10:
@umarcor Could you please help with creating that image for us and we could just update the CI after merging #27 ?

FaragElsayed2, 2023-04-09:
@atorkmabrains This issue has been addressed in [PV repo ](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pv)

umarcor, 2023-04-09:
@FaragElsayed2 I can't find any Dockerfile or reference to an OCI image in google/globalfoundries-pdk-libs-gf180mcu_fd_pv. Can you please point to the source?

atorkmabrains, 2023-04-09:
@umarcor We no longer use images. We use the conda-env maintained by @proppy.

What @FaragElsayed2 means that we no longer need this custom image anymore.

I'll close the issue and please reopen if you have any other concerns.

cc @proppy

closed 2023-04-09 by atorkmabrains