google/globalfoundries-pdk-libs-gf180mcu_fd_pr #53: Apply python linting and formatting rules to this repository

mithro, 2022-08-20:
A lot of the Python scripts are not complying with standard python formatting rules. We should enable tools which check the files are complaint.

Some tools which we should look at are;
 - `pep8`
 - `black`
 - `pylint`
 - Something which checks for trailing whitespace.
 - Something which checks for license headers.

Setting these up can be showing by looking at the F4PGA repositories.

mithro, 2022-08-20:
@umarcor - Can you help here?

atorkmabrains, 2022-08-21:
@mohanad0mohamed Could you please take a look here?

FaragElsayed2, 2023-04-09:
@atorkmabrains This issue has been addressed in [PV repo ](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pv)

atorkmabrains, 2023-04-09:
@mithro I think we can close this as that was addressed in the PV repo.

FaragElsayed2, 2023-04-09:
@atorkmabrains I have assumed that we will add new Pcell generators. And for regression used in models, linting issues have been fixed in Mabrains/PR repo.

atorkmabrains, 2023-04-09:
Thanks @FaragElsayed2 I could see the linting issues has been fixed in Mabrains and Efabless versions.

@mithro Could we close this issue?

atorkmabrains, 2023-04-09:
@mithro I have opened another issue for merging of Efabless updates #109 . 

I'll close this issue. Please don't hesitate to open if you think we still need this issue. As soon as we update Google version with Efabless, we should have the linting code here.

closed 2023-04-09 by atorkmabrains