google/globalfoundries-pdk-libs-gf180mcu_fd_pr #57: missing files in cells/ ?

proppy, 2022-09-08:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/cells currently only seem to contains klayout and xschem pcells.

In comparison the sky130 repo has a lot more:
https://github.com/google/skywater-pdk-libs-sky130_fd_pr/tree/main/cells
```
🍉 find skywater-pdk-libs-sky130_fd_pr/cells/ -type f | xargs -n1 basename | cut -f 2- -d '.'  | sort | uniq -c
     23 bins.csv
    270 cdl
    261 corner.spice
    261 data
    289 gds
    346 json
    289 magic.lef
     99 model.spice
    270 netlist.tsv
      1 out
     88 pm3.spice
    637 spice
    802 svg
```

should we have the equivalent for gf180mcu?

QuantamHD, 2022-09-08:
@atorkmabrains

atorkmabrains, 2022-09-09:
@QuantamHD We didn't split the model files. We kept it as is. It's better for the simulator. Not sure what is the request here.

mithro, 2022-09-09:
@atorkmabrains - We should prioritize humans and maintenance not software.

atorkmabrains, 2022-09-09:
@mithro Do you want us to split the files?