google/globalfoundries-pdk-libs-gf180mcu_fd_pr #58: Mismatch simulations run as default

lhrodovalho, 2022-10-07:
## Expected Behavior

Typical simulations using the spice header
`.include "../../../gf180mcuC/libs.tech/ngspice/design.ngspice"
.lib "../../../gf180mcuC/libs.tech/ngspice/sm141064.ngspice" typical`

should, as default, result in the same results each time the simulation is run.

## Actual Behavior

It uses mismatch parameters, so the simulation has random results (such as offset)

## Steps to Reproduce the Problem

1. Include the additional .param commands so the mismatch parameter flag is overwritten.

`.param
+  sw_stat_global = 0
+  sw_stat_mismatch = 0`

2. Change the local design.ngspice file manually.
 
## Specifications

- Version:
- Platform: Linux

QuantamHD, 2022-10-07:
Hi @lhrodovalho,

I'm not an expert on the simulation params. Do you know what we could do to fix this, or what might be causing the issue?

lhrodovalho, 2022-10-07:
Hi @QuantamHD

The problem is the design.ngspice file. The designer must include it alongside with the other lib files in the simulation testbench, as long as I know. It should be set to zero as default, not to one. Otherwise, the designer must override it in the testbench after including it. The sky130 open PDK mismatch param must be set outside the testbench. I didn't know this for GF180, because the simulations ran ok.

FaragElsayed2, 2023-04-09:
This issue has been addressed in [Mabrains/PR repo](https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/models/ngspice/design.ngspice)

atorkmabrains, 2023-04-09:
Hi @lhrodovalho This has been addressed in https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr 

which means it's fixed now in the latest volare release of the PDK. Please rerun and confirm. And open the ticket again if it's still there.

closed 2023-04-09 by atorkmabrains