google/globalfoundries-pdk-libs-gf180mcu_fd_pr #59: Resistor Model issue

christoph-weiser, 2022-10-09:
## Expected Behavior

ppolyf_u_2k_6p0 has a sheet resistance of 2k/square. 

## Actual Behavior

The resistance is much lower than this (68 ohm). 
This is also true for the other ppolyf resistors. Also the modelling of the parasitic capacitance
is not compatible with ngspice.

## Steps to Reproduce the Problem

```
* Netlist

Vin top GND 1
XR1 GND top GND ppolyf_u_2k_6p0 r_length=1e-6 r_width=1e-6 m=1

.control
op
let ires=-1*vin#branch
let res=1/ires
print res
.endc

.inc design.ngspice
.lib sm141064.ngspice res_typical

.GLOBAL GND
.end
```

mithro, 2022-10-09:
@atorkmabrains / @mkkassem - Thoughts?

atorkmabrains, 2022-10-10:
@mithro and @christoph-weiser We will update the resistors models in ng-spice to make sure that they match.

atorkmabrains, 2022-10-18:
@mithro and @christoph-weiser Please find the fix for this issue here: https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/61

christoph-weiser, 2022-10-18:
@atorkmabrains thanks for the update. I can confirm that the resistance is now corrected.

I'm however still wondering about the modelling of the substrate capacitance. 

``` * model for substrate capacitor (pF/um2)

.model fox_sub c cox=8.85E-05

c1  1 3  fox_sub       l='r_l/2' w=r_w dtemp=dtemp
c2  2 3  fox_sub       l='r_l/2' w=r_w dtemp=dtemp
``` 
From the ngspice manual i cannot find cox being a valid input to the semiconductor capacitor model or any of the other "c" models for that matter.

atorkmabrains, 2022-10-18:
@christoph-weiser There is a capacitance associated with the resistor model in the subcircuit definition for the resistor. I don't know if they have taken the substrate capacitance the way you mentioned. I also didn't quantify that against the silicon R/C data. We are planning to do that later.