google/globalfoundries-pdk-libs-gf180mcu_fd_pr #60: klayout deck location

proppy, 2022-10-14:
Is this repo the right place for the DRC and LVS rules? 

I find it hard to discover and feels more general than closely tied to the "primitive cells".

/cc @mithro @QuantamHD

mkkassem, 2022-11-07:
@proppy I placed them initially. I am thinking to move them under ./tech .. see below example for magic and netgen

https://github.com/RTimothyEdwards/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/open_pdks/tech

mithro, 2022-11-07:
I think @proppy was suggesting they might make more sense in the https://github.com/google/gf180mcu-pdk repo?

mkkassem, 2022-11-08:
@mithro that actually seems a better fit if what you mean is -  https://github.com/google/gf180mcu-pdk/tech

mithro, 2022-11-08:
Yeah, that seems reasonable?

FaragElsayed2, 2023-04-09:
@atorkmabrains This issue has been addressed in [PV repo](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pv)

atorkmabrains, 2023-04-09:
@proppy and @mkkassem We have moved DRC and LVS to pv repo. Do you want to move everything related to klayout in one repo?