google/globalfoundries-pdk-libs-gf180mcu_fd_pr #70: cells/klayout/pymacros/README: should document how to generate pcells

proppy, 2022-11-09:
## Expected Behavior

https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/cells/klayout/pymacros README should document how to generate gds using klayout and how to regenerate the tests in https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/cells/klayout/pymacros/testing.

## Actual Behavior

https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/cells/klayout/pymacros assume developers/contributors are familiar with the process to generate pcells using klayout and update testcases gds.

atorkmabrains, 2022-11-10:
@proppy and @mkkassem 

Please check the usage in this document:
https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/cells/klayout/pymacros/README.md

atorkmabrains, 2022-11-10:
@proppy About regression, please find more information here:
https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/cells/klayout/pymacros/testing/README.md

proppy, 2022-11-11:
@mkkassem @atorkmabrains I think you can easily create a separate pull request for the pcell changes this way:
```
# clone upstream repo
git clone https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr
cd globalfoundries-pdk-libs-gf180mcu_fd_pr
# add your fork remote
git remote add mabrains git@github.com:mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr.git
git fetch mabrains
# create feature branch
git checkout -b improve-pcell-docs
# checkout (only) pcells changes
git checkout mabrains/main cells/klayout/pymacros
# create a new collapsed commit for (only) pcells changes 
git commit -m 'cells/klayout/pymacros: improve documentation'
# push back to your fork
git push mabrains improve-pcell-docs
# create the pull request
```

atorkmabrains, 2022-11-11:
- [ ] @proppy I have discussed that with @mithro before. That can't be done. Primitive renaming is a lateral change that affects everything and if you take one part and leave the other parts of the PDK unchanged, it will break the PDK. And there is another issue, doing PR will not work as well as the amount of changes are massive. And github doesn't allow that. I have shown this to @mithro and @mkkassem. That's why I was asking you guys to get the primitives renaming PRs in first before we change anything.

proppy, 2022-11-11:
I think that if you:
1. take the instruction in https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/70#issuecomment-1311156971
2. force push to https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/32 branches (https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/primitives_names_part1)
3. mark it as fixing this issue

That should be enough to provide a coherent set of change to review and merge (that include both naming change and appropriate documentation update to evaluate it).

atorkmabrains, 2022-11-11:
@proppy I have all updates made in mabrains branch. I advise you to just replicate what you have here with what's in Mabrains.

proppy, 2022-11-11:
@atorkmabrains so you'd prefer me to do the changes highlighted in https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/70#issuecomment-1311156971 and https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/70#issuecomment-1311302640 rather than giving it a go?

atorkmabrains, 2022-11-11:
@proppy My advise is basically to replace what you have here with what's in Mabrains fork. And after that let review one by one. If there are any other changes required, we could handled them one by one. If you have doubts in my proposal, please go ahead and check the diff against what's in Mabrains. I'm not responsible of any issue if you didn't update to match Mabrains repo.

proppy, 2022-11-11:
> And after that let review one by one. If there are any other changes required, we could handle them one by one. 

I'd prefer for us to work with the pull request as I suggested. I will go ahead and update them then.

atorkmabrains, 2022-11-11:
@proppy Please do whatever you need at your end. But make sure that at the end that you match what I have on Mabrains side.

I believe you misunderstood me btw, what I'm saying is that you take what I have and push it to Google repo. Once that's done we could start adding PRs on Google side.

Please try to do that at your end.

atorkmabrains, 2023-04-10:
@FaragElsayed2 We need to follow up on this in efabless version.