google/globalfoundries-pdk-libs-gf180mcu_fd_pr #76: Klayout LVS should extract *mos_6p0 for 5V transistors

egorxe, 2022-11-21:
Currently Klayout LVS script extracts pmos_5p0 and nmos_5p0 models for 5V transistors (see [here](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/74e4ec59b55bcf5be2f153abff8519d15ebe21fa/rules/klayout/lvs/gf180mcu.lvs#L1302) for example). This is inconsistent with SPICE deck which uses *mos_6p0 for both 5V and 6V transistors. Probably LVS script should be changed a bit to extract 6p0 models always.

If this change is OK (no 5p0 models planed in the future), I'll create a pull request.

mithro, 2022-11-21:
@mkkassem & @atorkmabrains - Can you look into this?

egorxe, 2022-11-22:
@atorkmabrains Layers are different between 3.3V and 5/6V devices (Dualgate), but they are the same between 5V and 6V, only channel length is different. Anyway currently there is no *mos_5p0 models in SPICE deck, so it is impossible to perform LVS against known good SPICE netlist checked in simulation without netlist manipulations. So in my opinion either LVS scripts should be changed or *mos_5p0 should be aliased to *mos_6p0 in SPICE deck. There is no reason to close this issue until it's possible to pass LVS against simulated SPICE.

mithro, 2022-11-22:
I think I agree with @egorxe with 
> the *mos_5p0 should be aliased to *mos_6p0 in SPICE deck

atorkmabrains, 2022-11-22:
@mithro and @egorxe I need to review few things and get back to you.

egorxe, 2022-11-22:
I would also like to note that even with any of changes suggested above, KLayout LVS still will not pass out of the box. KLayout extracts devices as SPICE models, but transistors in netlists from schematics are usually expressed as subcircuits. Subcircuits are translated into comprehensive models inside SPICE library (such as sm141064.ngspice) which can't be parsed by KLayout. 

To pass KLayout LVS one will have to manually replace all transistor instances in schematics netlist from subcircuits (X*) to models (M*). Or do the reverse replacement in netlist extracted with KLayout and perform comparison in netgen. This could be done with python script for example. It would be very useful to document some recommended  way to perform LVS with small example, cause currently it could be very hard to comprehend it for a newcomer.

mithro, 2022-11-22:
@proppy - You where looking at LVS recently.

atorkmabrains, 2022-11-23:
@egorxe There is an option I believe in xschem that generate an "lvs" net list rather than simulation net list as I remember. BTW, it's an industry standard to have different net lists for different purposes.

egorxe, 2022-11-23:
@atorkmabrains I'm accustomed with Cadence and Mentor tools, thank you. And I'm not arguing with the idea of separate netlists. But xschem is not Virtuoso and KLayout (or Magic) is not Calibre. What I'm trying to say is that there should be documented and/or scripted way to pass LVS with current versions of open source EDA tools.

atorkmabrains, 2022-11-23:
@egorxe There is, we already have done a video tutorial for that. But it's not published till now. 

Mabrains is trying to get the following tools to a state very close to the commercial custom design tools. (Side note: I worked for Mentor for almost 9 years on custom design tools and verification tools):
- Xschem
- Klayout

We don't focus on any other tools for now as Mabrains. We actually have even a library manager internally that is similar to the commercial tool and integrates everything and makes everything smooth.

Anyhow, ping me if you need more info.

atorkmabrains, 2022-11-23:
@egorxe As I told you, there is a way to create a netlist that is LVS compatible in xschem directly. Here is how according to Stephan reply:

The first option seems already implemented. The Xschem SImulation menu has the entry "use spiceprefix attribute". All devices that use this attribute (for example all sky130 mosfets) will be  netlisted as X-subcircuits if the spiceprefix (usually set to 'X') is used in the symbol and the option to use it is enabled in the SImulation menu. I need to understand if there is a different requirement for this specific issue.

Cc @mithro

atorkmabrains, 2022-11-23:
https://github.com/StefanSchippers
Cc @stefanschippers

egorxe, 2022-11-23:
> @egorxe I never said that you need to do it by hand or custom script in xschem. Unfortunately, I just realized that xschem doesn't do that out of the box. But it should.

Great!

> The first option seems already implemented. The Xschem SImulation menu has the entry "use spiceprefix attribute". All devices that use this attribute (for example all sky130 mosfets) will be [...]

Thanks, this seems to be it. I failed to actually test this way cause if I try to generate SPICE netlist in xschem without this "Simulation-Use spiceprefix attribute" option in menu, this check magically restores itself. But this is xschem, not LVS issue anyway. I hope this procedure will be documented in your video and text documentation.

> I need to understand if there is a different requirement for this specific issue.

@atorkmabrains Problem described in issue name still stands. There are no *mos_5p0 devices in SPICE deck or xschem (even here https://github.com/StefanSchippers/globalfoundries-pdk-libs-gf180mcu_fd_pr). Either LVS should extract everything as *mos6p0 or *mos5p0 should be added to SPICE deck and xschem. The only difference between 5V and 6V MOS KLayout PCells seems to be the presence of V5_XTOR marking layer (defined as marking here https://gf180mcu-pdk.readthedocs.io/en/latest/physical_verification/design_manual/drm_04_1.html). If this V5_XTOR layer does not influence masks is there a reason to separate 5V & 6V transistors in LVS? May be it's simpler to just represent 5V in LVS as *mos_6p0 as already done in SPICE and xschem?

egorxe, 2022-11-23:
@atorkmabrains Then I suggest to keep this issue open until *mos_5p0 devices are added into SPICE deck and xschem. They can be just aliases for *mos_6p0. I've changed this issue name a bit to reflect it.

FaragElsayed2, 2023-04-09:
@atorkmabrains `mos_5p0 & mos_6p0` use same model but they are different in LVS as `mos_5p0` uses `dualgate & V5_xtor` marker layers but mos_6p0 uses `dualgate` layer only. So we need to use mos_5p0 devices in LVS extraction as mentioned in DRM.

[image]
[image]
[image]

atorkmabrains, 2023-04-09:
@mithro @mkkassem @RTimothyEdwards We need to reach out to GF for this.

StefanSchippers, 2023-04-09:
`nfet_05v0.sym` can be added and can refer to the same spice model as `nfet_06v0.sym`.
The netlisting rule can be differentiated to accomodate LVS checking, while keeping the same netlist for spice sims.
xschem has 2 options (simulation menu) that are mostly used for LVS
- disable the spiceprefix (X) attribute so devices should be netlisted as `Mxxxx` instead as `XMxxxx`
- set LVS netlist. This will keep the top level wrapped inside a .subckt as usually happens with CDL netlists. When LVS netlist is set transistor symbols will use `lvs_format` attribute as the netlisting rule (if existing) instead of `format` to generate the netlist. This is already used to avoid generating the ad/as/pd/ps ... parameters that are not used in LVS.
We just need to know how Klayout extracts nfet_05v0 and nfet_06v0 to produce a matching netlist from xschem.

The above spiceprefix and LVS netlist flags can be set also from command line if we want to generate netlists from scripts, by setting tcl variables `spiceprefix` and `lvs_netlist`.

atorkmabrains, 2023-04-09:
@StefanSchippers That would be a great update. Could you please do that and create PR on: https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr for this?

atorkmabrains, 2023-04-09:
I still think that we need to reach out to GF for this. 

@mithro @mkkassem @RTimothyEdwards

closed 2023-06-12 by atorkmabrains