google/globalfoundries-pdk-libs-gf180mcu_fd_pr #77: Is there a layer map file?

growly, 2022-11-22:
I need a straightforward mapping of layer names to GDS types/subtypes. I believe this information is captured by various tool configuration files, like `gf180mcu.lyp` for klayout, but it would be helpful to have the source in order to produce another tool-specific configuration file instead.

For sky130 such a file was available in non-public versions, so I'm curious if there's a similar situation for this PDK.

mithro, 2022-11-22:
There was also the [SKY130 (SkyWater PDK) -- Layer (GDS-II / Magic / Calibre / etc) Information spreadsheet for SKY130](https://docs.google.com/spreadsheets/d/1oL6ldkQdLu-4FEQE0lX6BcgbqzYfNnd1XA8vERe0vpE/edit#gid=459375686).

atorkmabrains, 2022-11-22:
@growly Please find it here:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/tech/klayout/gf180mcu.lyp

cc @mithro

mithro, 2022-11-22:
@atorkmabrains - @growly mentioned he had found the `gf180mcu.lyp` file but was looking for something a bit easier to process.

growly, 2022-11-23:
Yes I was hoping for a simpler format, such as the source of whatever produced the `.lyp` in the first place (a spreadsheet would be nice). However I could extract it from the `gf180mcu.lyp` file or the magic file, if someone could please clarify what this means?

1. ` <source>1/222@1</source>`
1. `<source>Poly2 30/0@1</source>`

Example (1) doesn't have a layer name?

The numbers are layer/datatype@something? What is the last number?

atorkmabrains, 2022-11-23:
@growly I'm sorry misread what you need. Here is a CSV file that is used for documentation. You could use pandas to read this file in Python for example:

https://github.com/google/gf180mcu-pdk/blob/ff2a6056a2e69eed4da7a13c7a050cdbde6b7b0c/docs/physical_verification/design_manual/tables_clear/06_Drawn_layer10.csv