google/globalfoundries-pdk-libs-gf180mcu_fd_pr #79: consider using a test framework

proppy, 2022-11-29:
## Expected Behavior

Tests should be using [python testing best practices](https://docs.python-guide.org/writing/tests/).

One popular framework is `pytest`: https://docs.pytest.org/en/7.2.x/ as it comes with:
- automated test discovery
- and a rich ecosystem of plugins: https://docs.pytest.org/en/7.2.x/reference/plugin_list.html#plugin-list (including things like https://pypi.org/project/pytest-xdist/ that allow multiple tests to be run concurrently).

## Actual Behavior

Tests uses monolytics python scripts with manual `exit()` invocation, see https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/rules/klayout/drc/testing/run_regression.py#L243

FaragElsayed2, 2023-04-26:
@atorkmabrains Did we hit this target in the new testing environment we have in PV repo ?

atorkmabrains, 2023-05-01:
@FaragElsayed2 I don't believe so. I believe we still use the normal regression flow for PV. I recommend that we open another issue on PV repo and close this one.

FaragElsayed2, 2023-05-02:
@atorkmabrains Should I do that or wait for @proppy to move it ?

atorkmabrains, 2023-05-02:
@FaragElsayed2 Please open the other issue on the PV repo and we I'll close that one. Please make sure to link the new issue to this issue.

FaragElsayed2, 2023-05-02:
@atorkmabrains Done

atorkmabrains, 2023-05-02:
@FaragElsayed2 Could you please link it here?

FaragElsayed2, 2023-05-02:
Refer to https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pv/issues/76

Cc @atorkmabrains

atorkmabrains, 2023-05-02:
@proppy I'm closing this issue as we moved it to PV repo as @FaragElsayed2 has highlighted above.