google/globalfoundries-pdk-libs-gf180mcu_fd_pr #80: Missing 5V transistor in transistor models set for GF180 MCU

azwefabless, 2022-12-05:
## Expected Behavior
IP is drawn with 5V device and has 5V .libs and device modesl were expected to hav 5V devices.

## Actual Behavior
Actual device models present are: 
* Models included in this release :
*
*      ModelName          Description
*      ---------          -----------
*      nmos_3p3           Subcircuit model for 3.3V NMOS
*      pmos_3p3           Subcircuit model for 3.3V PMOS
*      nmos_6p0           Subcircuit model for 6.0V NMOS
*      pmos_6p0           Subcircuit model for 6.0V PMOS
*      nmos_3p3_sab       Subcircuit model for 3.3V NMOS with Drain side SAB
*      pmos_3p3_sab       Subcircuit model for 3.3V PMOS with Drain side SAB
*      nmos_6p0_sab       Subcircuit model for 6.0V NMOS with Drain side SAB
*      pmos_6p0_sab       Subcircuit model for 6.0V PMOS with Drain side SAB
*      nmos_6p0_nat       Subcircuit model for 6.0V native NMOS

## Steps to Reproduce the Problem
Open file: 
https://raw.githubusercontent.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/main/models/ngspice/sm141064.ngspice
Read header.

azwefabless, 2022-12-05:
This seems to be related to issue #76 but we should be clear that this is explicitly a blocker for Booting a chip at power up. Aliasing the models to the 6p0 model is a crude hack that may be sufficient to provide some function.

mithro, 2022-12-05:
@azwefabless Are you planning on sending a pull request for this?

azwefabless, 2022-12-05:
Trying to work through options internally but not confident in the POR simulations yet using the 6p0 transistors after looking at the model differences. The same issue seems to be present in the source data we got from GF.  (Was not created in the translation to open source).

azwefabless, 2022-12-08:
The answer received from GF support last night clarifies that they expect the 6p0 model to be used for the 5p0 device.

Begin Quote:
"The 5V transistors leverage the 6V device models. Please see the picture below for the details."
[image]

azwefabless, 2022-12-08:
Prior post was submitted with GF permission.

FaragElsayed2, 2023-04-26:
@atorkmabrains Could we close this one ?

atorkmabrains, 2023-05-01:
@azwefabless Will close this ticket as no further action is required.