google/globalfoundries-pdk-libs-gf180mcu_fd_pr #89: Warnings from the `actions/setup-python@v3` GitHub Action on CI

mithro, 2023-01-12:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/actions/runs/3904358547/jobs/6670037402
```
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Successfully setup CPython (3.9.16)
/opt/hostedtoolcache/Python/3.9.16/x64/bin/pip cache dir
/home/runner/.cache/pip
Warning: The `save-state` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Warning: The `save-state` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
pip cache is not found
Warning: The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
```

FaragElsayed2, 2023-04-26:
@atorkmabrains I think this issue is no longer exist in [PV repo](https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv)

You could find CI [here](https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv/actions/runs/4808613757/jobs/8558810058).

FaragElsayed2, 2023-04-26:
@atorkmabrains Not yet

atorkmabrains, 2023-05-01:
@FaragElsayed2 When we will integrate those changes in Google version?

FaragElsayed2, 2023-05-02:
@atorkmabrains Once we have LVS in Google version.

atorkmabrains, 2023-05-02:
@FaragElsayed2 How many tables did we get so far in Google? Could we add more for DRC to finish it?

FaragElsayed2, 2023-05-02:
@atorkmabrains We have added about 12 table. I'm waiting for merging opened PRs before adding more.

atorkmabrains, 2023-05-02:
Thanks @FaragElsayed2