google/globalfoundries-pdk-libs-gf180mcu_fd_pr #96: Issue with GF180 PDK and symbols import into Xshem and incompatibility with definitions in sm141064.spice

spnadig, 2023-01-31:
## Expected Behavior
ngspice needs to run with no errors

## Actual Behavior
error:
[image]

##NETLIST
** sch_path: /usr/local/google/home/nigelcoburn/MixedSignal/Designs/TopLevel_oscillator.sch
**.subckt TopLevel_oscillator
X1 VDD GND Vout Vin Buffer
Vdd VDD GND 1.8
.save i(vdd)
Vin Vin GND pulse(0 1.8 1ns 1ns 1ns 4ns 10ns)
.save i(vin)
**** begin user architecture code

blabla

.tran 0.01n 1u
.saveall

name=TT_MODELS1 only_toplevel=false
.include /usr/local/google/home/nigelcoburn/MixedSignal/silicon-env/share/pdk/gf180mcuC/libs.tech/ngspice/design.ngspice
.lib /usr/local/google/home/nigelcoburn/MixedSignal/silicon-env/share/pdk/gf180mcuC/libs.tech/ngspice/sm141064.ngspice typical

**** end user architecture code
**.ends

* expanding   symbol:  Designs/Buffer.sym # of pins=4
** sym_path: /usr/local/google/home/nigelcoburn/MixedSignal/Designs/Buffer.sym
** sch_path: /usr/local/google/home/nigelcoburn/MixedSignal/Designs/Buffer.sch
.subckt Buffer VP VN Y A
*.iopin VP
*.iopin VN
*.ipin A
*.opin Y
X1 VP VN net1 A Inverter
X2 VP VN Y net1 Inverter
.ends

* expanding   symbol:  Designs/Inverter.sym # of pins=4
** sym_path: /usr/local/google/home/nigelcoburn/MixedSignal/Designs/Inverter.sym
** sch_path: /usr/local/google/home/nigelcoburn/MixedSignal/Designs/Inverter.sch
.subckt Inverter VP VN Y A
*.ipin A
*.iopin VP
*.iopin VN
*.opin Y
XM3 Y A VP VP pfet_03v3 L=0.28u W=0.22u nf=1 ad='int((nf+1)/2) * W/nf * 0.18u' as='int((nf+2)/2) * W/nf * 0.18u'
+ pd='2int((nf+1)/2) * (W/nf + 0.18u)' ps='2int((nf+2)/2) * (W/nf + 0.18u)' nrd='0.18u / W' nrs='0.18u / W'
+ sa=0 sb=0 sd=0 m=1
XM4 Y A VN VN nfet_03v3 L=0.28u W=0.22u nf=1 ad='int((nf+1)/2) * W/nf * 0.18u' as='int((nf+2)/2) * W/nf * 0.18u'
+ pd='2int((nf+1)/2) * (W/nf + 0.18u)' ps='2int((nf+2)/2) * (W/nf + 0.18u)' nrd='0.18u / W' nrs='0.18u / W'
+ sa=0 sb=0 sd=0 m=1
.ends

.GLOBAL GND
.GLOBAL VDD
.end

## Steps to Reproduce the Problem

[image]

## Specifications

- Version:
- Platform:

proppy, 2023-01-31:
@msaligane did you have it for OpenFASoC?

spnadig, 2023-01-31:
edit -  to be clear, the error persists after deleting line blala from netlist. the issue is we cannot find symbols for devices  example - pmos3p3 in our components-> /devices which is consistent with  sm141064.spice

mithro, 2023-01-31:
Could this be related to https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/6 ?

msaligane, 2023-01-31:
@proppy We do not use a schematic editor in the normal openfasoc flow. 

If it is regarding "blabla", yes, my student encountered this error before, but it was resolved. If there is a syntax error while specifying the ngspice model path this happens.

spnadig, 2023-02-01:
@proppy - this is the path we use 
[image]

@msaligane  - we cannot find pfet03v3 nfet03v3 under components ->/devices like you do in your designs. Did you do something specific in your PDK imports to xschem? 
Our 0_top.sch has test_pfetxxx and these symbols have no terminals and I cannot use them in my schematics[image][image]

msaligane, 2023-02-01:
Are you using an old version of the PDK?
The naming rule is different, in that repo, check `pmos_3p3` and `nmos_3p3`.

We are using the latest version installed by volare. This is the version of gf180pdk:
`b8c6129fb60851c452a3136c2b8c603bb92cb180`

@madhulikal @StanleyAlwaysLoveYou - can you please help? Thanks!

proppy, 2023-02-01:
@spnadig can you print the result of:
```
conda list | grep open_pdks
```

spnadig, 2023-02-01:
@proppy - 
[dump.txt](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/files/10552316/dump.txt)

madhulikal, 2023-02-01:
> @proppy - this is the path we use [image]
>
> @msaligane - we cannot find pfet03v3 nfet03v3 under components ->/devices like you do in your designs. Did you do something specific in your PDK imports to [...]

@spnadig - This syntax should help resolve the issue without manually deleting 'blabla' everytime.
format="tcleval(@value )" value=".lib and .include model filepaths".  

Also, can you check if the path of device models in xschemrc file is referencing to the correct pdk?
In older version too, if the xschemrc file is correct, this error should be resolved.

proppy, 2023-02-01:
I think the path to the model in `xschemrc` needs to be fixed with:
```
sed -i -e 's/ngspice\/models/gf180mcuC\/libs.tech\/ngspice/'   ~/silicon-env/share/pdk/gf180mcuC/libs.tech/xschem/xschemrc 
```

proppy, 2023-02-07:
Fixed with https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/91

closed 2023-02-07 by proppy