google/globalfoundries-pdk-libs-gf180mcu_fd_pr #101: lvs script should be able to import xschem netlist

proppy, 2023-02-06:
By default xschem extract netlist with the `.spice` extension and include modeling equation as part of the parameters (which klayout can't parse).

It would be nice if the lvs script was trying multiple extension and simplifying the netlist prior to parsing.

StefanSchippers, 2023-02-06:
If needed this can be easily done in xschem. 
for each mos symbol add a `lvs_format` attribute in addition to the format attribute that is used for simulation netlists:
`lvs_format="@spiceprefix@name @pinlist @model L=@L W=@W nf=@nf m=@m"`
This will generate in the LVS netlist (Simulation->LVS netlist is checked) only the parameters that are relevant for LVS:

`XM1 D G S B nfet_03v3 L=0.28u W=0.22u nf=1 m=1`

proppy, 2023-02-06:
@StefanSchippers that's awesome! (and would be way cleaner that removing the argument as part of the LVS script like #99 is attempting to do)

proppy, 2023-02-06:
@atorkmabrains @mithro we should consider adding `lvs_format` attribute to the primitive symbols in https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/cells/xschem/symbols

atorkmabrains, 2023-02-07:
@proppy yes

PR #102 (2023-02-07): Improve transistor definitions in xschem lvs export
- strip unusted parameters on LVS export
- avoids X prefix
- fix substrate pin of vnpn_10x10.sym (was named "B" instead of "S")

 (Still to be decided if we want to make similar changes for LVS on other devices (moscap and others)

Fixes #101
closed 2023-02-07 by proppy