google/globalfoundries-pdk-libs-gf180mcu_fd_pr #104: can LVPWELL be used without DNWELL?

proppy, 2023-02-07:
Looking at the current pcells, it seems that drawing the `LVPWELL` for nfet:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/11770e5e5bae2a2072899e7ef282c080e6c5d762/cells/klayout/pymacros/cells/draw_mos.py#L192-L195
is gated by having `deepnwell` param enabled:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/11770e5e5bae2a2072899e7ef282c080e6c5d762/cells/klayout/pymacros/cells/draw_mos.py#L172

Are those two always tied together? looking at the design manual there seems to be a valid use case for using `LVPWELL` outside of `DNWELL`:
https://gf180mcu-pdk.readthedocs.io/en/latest/physical_verification/design_manual/drm_07_04.html

>  If this layer is used without DNWELL (outside DNWELL), the body of all those transistors will by default be connected to P-substrate potential.

RTimothyEdwards, 2023-02-07:
@proppy :  Yes, LVPWELL outside of DNWELL is equivalent to substrate and should be treated as such.  That usage is in much of the GF IP.

atorkmabrains, 2023-02-07:
Thanks @RTimothyEdwards

proppy, 2023-02-07:
@RTimothyEdwards Thanks for the explanation.

@atorkmabrains would it makes sense to update the pcells to always draw `LVPWELL` when `Bulk Tie` is set?
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/11770e5e5bae2a2072899e7ef282c080e6c5d762/cells/klayout/pymacros/cells/draw_mos.py#L192-L195

spnadig, 2023-02-08:
@msaligane - this was the issue I was talking about

proppy, 2023-02-15:
@atorkmabrains currently it seems that `LVPWELL` is only drawn when `DNWELL` is set in the pcell option, would it make sense to update the pcell implementation to always draw `LVPWELL` when bulk tie is enabled independently of the state of the `DNWELL` option?

atorkmabrains, 2023-02-15:
@proppy That's a separate issue. `LVPWELL` is optional or don't care in case that `DNWELL` is not drawn. If you draw, `DNWELL` you can draw `LVPWELL`. Pcell is functioning correctly, no need to add this modification into the pcell.

@spnadig Could you please show me the design that caused this?

@spnadig I advise that you use DRC and LVS from https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr? Also, please stay tuned as we are going to move LVS and DRC to: https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv.

cc @msaligane

atorkmabrains, 2023-02-15:
> @RTimothyEdwards Thanks for the explanation.
>
> @atorkmabrains would it makes sense to update the pcells to always draw `LVPWELL` when `Bulk Tie` is set? [...]

No, as I mentioned in my comment above, `LVPWELL` is totally ignored in everything if there is no `DNWELL`. No need to modify the pcell. 

The issue that @spnadig is a separate issue. I need to look into it.

proppy, 2023-02-15:
My understanding is that in https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/104#issuecomment-1421127633 @RTimothyEdwards stated that `LVPWELL` without `DNWELL` was a valid usecase.

Shouldn't we make sure that this is also supported by the pcell?

atorkmabrains, 2023-02-15:
@proppy yes, you could draw `LVPWELL` but the correct action is to **not** to draw it by default as it's disregarded in case the `DNWELL` is not drawn. The pcell is designed to allow the `real` use case of the `LVPWELL`. I don't advise to add `LVPWELL` if you are not really using it. 

An exception to my comment above is the standard cells, standard cells should have `LVPWELL` just in case the user might want to use `DNWELL` later around the digital block later for any electrical reasons. But this use scenario is **only** valid for standard cells.

atorkmabrains, 2023-02-15:
@proppy I'm more concerned about the issue the @spnadig regarding the LVS. I need to understand that use case and why it didn't pass without `DNWELL` even if the `LVPWELL`. This has nothing to do with the pcell adding `LVPWELL`.

proppy, 2023-02-15:
@atorkmabrains I think this is due to the substrate connection of the bulk tie creating an additional signal (`gf180mcu_gnd`) when the `LVPWELL` is not present.

Is there a usecase for setting `Bulk tie` without drawing `LVPWELL`?

atorkmabrains, 2023-02-15:
@proppy `bulk tie` doesn't need to `LVPWELL` to happen, please read here:
https://en.wikipedia.org/wiki/CMOS

And BTW, that's the normal in almost all CMOS processes.

`GF180MCU` is a triple well technology, hence we need a `PWELL` marker:
https://www.researchgate.net/figure/Triple-well-NMOS-transistor-structure-See-the-colors-in-the-online-version_fig2_303530652

proppy, 2023-02-15:
> bulk tie doesn't need LVPWELL to happen, please read here: https://en.wikipedia.org/wiki/CMOS

Thanks for providing the links, I could find any mention of `bulk tie` nor its relation (if any) to `PWELL` on it, do you have an alternative source of documentation?

> GF180MCU is a triple well technology, hence we need a PWELL marker:
https://www.researchgate.net/figure/Triple-well-NMOS-transistor-structure-See-the-colors-in-the-online-version_fig2_303530652

Without a `PWELL` it seems that the current LVS extraction is not able to infer the substrate connection of the `BULK` tied to `SRC` terminal for nmos device effectively leaving it floating with a different ground net, is that intentional? Or is there a different way to provide a hint for this implicit connection in the layout?

atorkmabrains, 2023-02-15:
> Without a `PWELL` it seems that the current LVS extraction is not able to infer the substrate connection of the `BULK` tied to `SRC` terminal for nmos device effectively leaving it floating with a [...]

If what you mentioned is true that LVS requires `LVPWELL` to get the connectivity to substrate, then that's definitely a bug in LVS. I need the test case that has that issue. Also, are they using Google repo? If yes, then please tell them to use efabless one as Google repo is not updated and not cleaned.

atorkmabrains, 2023-02-15:
> > bulk tie doesn't need LVPWELL to happen, please read here: https://en.wikipedia.org/wiki/CMOS
> 
> Thanks for providing the links, I could find any mention of `bulk tie` nor its relation (if any) to `PWELL` on it, do you have an alternative source of documentation?

As for your question about this, you need to know more about micro-electronics `bulk tie` has many different words, you don't necessarily need to find in the link I provided. Here is a few good courses you could start looking at:
- https://ocw.mit.edu/courses/6-012-microelectronic-devices-and-circuits-fall-2005/pages/lecture-notes/
- http://bwrcs.eecs.berkeley.edu/Classes/icdesign/ee240_sp12/lectures.html