google/globalfoundries-pdk-libs-gf180mcu_fd_pr #107: DRC PL.2_3.3V mixes gate length rules with gate width rules

andylithia, 2023-02-17:
## Expected Behavior

W=0.22um, L=0.28um (the minimal PCell Size) shouldn't fail DRC

## Actual Behavior

[image]

## Steps to Reproduce the Problem

- Version: 11770e5e5bae2a2072899e7ef282c080e6c5d762
1. Instantiate a PCell with the default parameters
2. Run Klayout DRC

## Possible Cause

Rule PL.2_3.3V in gf180mcu.drc doesn't seem to check width and length separately (i.e. it assumes device width is always larger than length)

## References

[gf180mcu PDK Physical Verification](https://gf180mcu-pdk.readthedocs.io/en/latest/physical_verification/design_manual/drm_07_08.html#poly2)

@msaligane

proppy, 2023-02-17:
/cc @atorkmabrains

FaragElsayed2, 2023-04-09:
@atorkmabrains This issue has been addressed in [Efabless/PV repo](https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv)

atorkmabrains, 2023-04-09:
Thanks @FaragElsayed2

closed 2023-04-09 by atorkmabrains