google/globalfoundries-pdk-libs-gf180mcu_fd_pr #110: nfet_03v3_dw missing in PDK?

spnadig, 2023-04-22:
## Expected Behavior
In the ngspice device list under/share/pdk/ngspice/libs.tech/sm141064.ngspice we need to see seperate devices for nfets with and without DNWELL namely nfet03v3_dw and nfet_03v3 as shown in the device listing from GF. adding DNWELL needs to change the model of the nfet and point to a different model param in sm14064.ngspice

[image]
## Actual Behavior
[image]

## Steps to Reproduce the Problem
However, sm14064.ngspice does not seem to have seperate models for nfet_03v3_dw as expected. I only see nfet_03v3 in my pdk files
[image]

There are 2 issues here:
1. the names for the devices dont match with GF -so it is making it difficult to compare across platforms and tools. For instance magic would have issues reading these names- https://github.com/RTimothyEdwards/magic/issues/239
2. Some devices and its models are missing in this PDK - specifically the DNWELL nfet device. As a result the extraction of a DNWELL device seems to be mapping to a typical nfet which yields incorrect results

## Specifications

- Version:
- Platform: proppy/conda

spnadig, 2023-04-22:
@proppy

FaragElsayed2, 2023-04-26:
@spnadig 

For 1st issue, you could find the naming convention for standard cells and primitive devices [here](https://docs.google.com/spreadsheets/d/1G88IhyqXGVu-BgH8hA-_N2YQnNiyPoPEqvXSOzmOaaY/edit#gid=1803291424).

And please use this [documentation](https://gf180mcu-pdk.readthedocs.io/en/latest/physical_verification/design_manual/drm_15.html) instead one you have.

For 2nd issue, As mentioned in docs, we don't have separate model for deep nwell devices. It uses same model for typical ones.

Example: `nfet_03v3 & nfet_03v3_dn` uses `nfet_03v3` model.

I'm not sure if we should ask the foundry about this.

Cc @atorkmabrains

atorkmabrains, 2023-04-26:
Thanks @FaragElsayed2 

@spnadig Keep in mind that Deep Nwell devices are usually using the same models as far as simulation goes. I'll close this ticket as it's not a true issue.

closed 2023-04-26 by atorkmabrains